mfactory-osaka/ESPTimeCast #1: [Suggestion] use ESPhome

stephenlee-asana, 2025-06-21:
This looks pretty great and a potential, straightforward conversion to ESPhome!  One awesome advantage of doing so would be to enable compatibility with other microcontrollers like the ESP32.  In turn, this would easily enable something an ESP32 peripheral like bluetooth_proxy for something like Home Assistant.

WDYT?

wQQhce2g93Ei, 2025-06-22:
I had the same thought the minute I saw this. I would love to have this and manage it like I do all the other esphome devices I have. It makes for easy updates/modifications if you're already in the esphome world.

briis, 2025-07-07:
I have a minimal working version done now for ESPHome that uses an ESP32 device. I will work a bit more on it, and then post the yaml file here.

briis, 2025-07-09:
I am waiting for @mfactory-osaka to implement some new stuff before releasing this, but if anyone want to test the current version you can find it here on my fork of this project. Once @mfactory-osaka is ready with the new Arduino version I will try an implement as much as possible to the ESPHome version.

[Here is the link to my fork](https://github.com/briis/ESPTimeCast/tree/main/ESPHome). Please go through the README file on how to set it up.

The current implementation does not require Home Assistant to work, it runs in the same way as the Arduino project. I have started work on a version that will pull weather data from Home Assistant, making the call to OpenWeatherMap from the ESP device unnessecary.

trey504, 2025-08-09:
> I am waiting for [@mfactory-osaka](https://github.com/mfactory-osaka) to implement some new stuff before releasing this, but if anyone want to test the current version you can find it here on my [...]

Thanks. It works great on ESP Home. I just had to just remove one thing "esphome::to_string"

briis, 2025-10-01:
The Warning on the GPIO5 strapping pin, is just that. It is a warning and can safely be ignored. If you use a different board where the CS pin can mapped to another GPIO that does not create the warning, you can do that. This is just what I use for the current board.

In the fork on [My Github](https://github.com/briis/ESPTimeCast/tree/main/ESPHome) you can use the file `esp32-time-cast.yaml`, that will compile with the latest version of ESPHome without errors. Just remember to make the changes that fit to you.

closed 2025-10-01 by mfactory-osaka