mfactory-osaka/ESPTimeCast #8: Clock Duration/Weather Duration not change.

RadekKrepela, 2025-07-08:
After saving, this value will not change.

[image]

[image]

[image]

RadekKrepela, 2025-07-08:
Latest from today. 

If i chage vaule in Config.jsom and load by littlels it works ,but after any save by web i have defaoult value to 10000/5000

mfactory-osaka, 2025-07-08:
Updated so ill mark it resolved ;)

closed 2025-07-08 by mfactory-osaka