mfactory-osaka/ESPTimeCast #22: Character mixup in weather description

lome64, 2025-08-07:
I observed problems in the display of the weather description.
Setup: D1 Mini (clone, I guess) with 5V at the display.
I hoped the direct power supply to the display would fix the issue, but it didn't.

Sometimes, maybe once in 10 minutes, the text of the weather description is broken.
Strange characters are mixed into the text. Sometimes leftovers for other strings like "WiFi" and
sometimes irregular characters or bit patterns.

That only happens on scrolling. I never saw a problem on short, static messages.

I modified the source so it always prints the "[WEATHER] Description used: %s\n", weatherDescription.c_str() 
message to the serial console every time the display changes from clock to weather and found that  
weatherDescription.c_str()  is always ok. 
Also I monitored the text in desc.c_str() just before scrolling. That one is also always ok.
The error might be in the P.displayScroll() function itself.

I see that problem on all my 4 clocks. Independend of any language settings and such.

lome64, 2025-08-07:
I did some quick tests:
Phone close to the clock (50 cm): Errors
Phone switched off: Everything is fine
It's not the code - it's the electrical sensitive ESP board. 

I'll do some experiments with shielding and blocking caps in the next days and report.
Also I'll try to get a different ESP board. These here are CHEAP! ;-)

Thank you!

mfactory-osaka, 2025-08-07:
Wow, nice to know.
i'll close this for now but please report any findings if you don't mind.

My ESP32 boards were very cheap and they usually act up so I solder some pin headers to the EN pin and they kinda work better that way, don't know if might be related with your problem but maybe worth testing ;)

lome64, 2025-08-07:
Ok - it's fixed!

2 220nF caps soldered directly to pins 11 and 17 of the ESP8266 to ground helped a bit, but not that much.
But wrapping the entire board (except the antenna) in capton tape for insulation and then in grounded copper foil 
fixed the problem. 
No errors anymore, even if the phone with WiFi switched off, so it uses LTE+, placed 10cm away playing online video, 
did not disturb the clock in any way! 
Success!

I will go and wrap the other ESPs now.

mfactory-osaka, 2025-08-19:
I had the same problem when another esp board was near the active board, the documentation and all my research indicated that the electromagnetic interference was not to blame, but there I was experiencing the same problem as you. Anyway it seems that the interference was causing a shift on the variable memory buffer, so i added a fixed buffer to the variable so it cannot shift so I haven't seen the problem since then.

code is updated but not uploaded yet, I was going to add Air Quality Index to the device when I started having this problem, now that is resolved I don't know if adding AQI is something that people will use lol!