mfactory-osaka/ESPTimeCast #32: feature req: more gentle transition

y8s, 2025-08-30:
When the display finishes scrolling the weather, the screen is blank and then the day and clock pop in all at once.  it can be a bit distracting. when the weather text enters, it's more subtle because it's scrolling in.

Can the time and day also scroll in or appear more slowly some other way (reverse dissolve, brightness increase --though this doesn't work if I'm already running low brightness)

thanks for considering!

closed 2025-08-31 by mfactory-osaka

y8s, 2025-09-04:
so I did run this and it feels better but it's so subtle! what did you change?

mfactory-osaka, 2025-09-04:
Yeah it's a very subtle change.
Before after every scrolling screen like "weather description" or "countdown" the device will change immediately back to clock, now it scroll in after such transition ;)