mfactory-osaka/ESPTimeCast #33: [Suggestion] Do not retrieve wifi credentials in APMode after initial setup

lexmong, 2025-09-06:
Saved credentials are shown in APMode if the clock reboots and can't connect to the saved network. I don't think most users change the APMode's default SSID/password. Anyone googling the default SSID can get the default password from the repo.

mfactory-osaka, 2025-09-06:
I don't really understand what you mean by that, can you elaborate?

lexmong, 2025-09-06:
So the user does the first-time setup and saves his home wifi credentials via the AP mode with the default SSID and password ESPTimeCast:12345678

When the ESPTimeCast reboots, it tries to connect to the home wifi. If for some reason it fails to connect, it goes into AP mode and still uses ESPTimeCast:12345678

If you connect to the ESPTimeCast and open http://192.168.4.1, you can now see the user's home network SSID and password in the text fields.

You should leave the SSID and password fields blank if the device is in AP mode so it does not leak the user's network credentials.

PS. I don't actually own the device; I just saw "ESPTimeCast" in the available networks of my pc, googled it, and landed on this repo. Tried the default password and I can now see my neighbor's SSID and password (I advised him to change his password).

lexmong, 2025-09-06:
He did get it to work for a few days. But we did have a power outage, so my guess is that when the power came back, the esp booted before the router and after 30 seconds it got into AP mode.

If for some reason WiFi.status() returns something else than WL_CONNECTED at boot (router out of range, or any other error) it will go in AP mode with the default SSID and password, and the fields on the interface will contain the user's credentials if first-time setup was completed.

Anyway, nice project, will probably build one too, but I will change the default password :smiley:

lexmong, 2025-09-06:
Yes the web UI

I just tried on an ESP8266 I got lying around and got the same result. 

Here are the steps to reproduce:
1. Setup wifi credentials via web UI
2. Wait for the device to reboot and get an IP
3. Unplug the router
4. Reboot the ESP8266
5. Connect to ESPTimeCast with default credentials and open the web UI
6. The SSID and password fields are filled with the wifi credentials

mfactory-osaka, 2025-09-06:
Got it, I think that the device is keeping the old credentials when starting as AP, I will implement a fix. Thanks for providing information about this.

I will let you know when the fix is implemented so you can give it a try ;)

mfactory-osaka, 2025-09-07:
Updated ;)

closed 2025-09-07 by mfactory-osaka