mfactory-osaka/ESPTimeCast #39: [Suggestion] Adding Russian language

Mart-Black, 2025-09-21:
Many of my friends from Russia use this watch. Is there a way to add Russian language support? Thanks :)

mfactory-osaka, 2025-09-22:
The openweather map doesnt support russian language :(
If the language ever becomes available in the API i will add it.

mfactory-osaka, 2025-09-22:
Sorry i was looking at an outdated API information, Russian is supported so yeah, I will add Russian on the next update. Cyrillic will be transcripted to latin so "cloudy" will appear as "oblachno"

How about the days of the week? Do you use 3 letters like MON for monday? What will be the Russian equivalent?

Mart-Black, 2025-09-22:
In Russian, the days of the week are displayed in two letters.
Below it starts from Sunday, Monday, etc.
{ "ru", { "в&с", "п&н", "в&т", "с&р", "ч&т", "п&т", "с&б" } }, 
Months are displayed in three letters.
{ "ru", { "я&н&в", "ф&е&в", "м&а&р", "а&п&р", "м&а&й", "и&ю&н", 
          "и&ю&л", "а&в&г", "с&е&н", "о&к&т", "н&о&я", "д&е&к" } },
To count down the days, hours, minutes, seconds, I asked AI to help with declensions.
Therefore, I came to the conclusion that you can simply shorten the words.

String currentSegmentText = String(days) + " " + (days == 1 ? "ДЕНЬ" : "ДНЕЙ");
...
sprintf(buf, "%02ld ЧАС", hours);
...
sprintf(buf, "%02ld МИН", minutes);
...
sprintf(buf, "%02ld СЕК", seconds);

if (!label.startsWith("ДО:") && !label.startsWith("до:")) {
    label = "ДО: " + label;
}

But if you want to make days, hours, minutes, and seconds declinable, then that's what you've achieved from AI.

String plural_ru(int n, String form1, String form2, String form5) {
  n = abs(n) % 100;
  int n1 = n % 10;
  if (n > 10 && n < 20) return form5;
  if (n1 > 1 && n1 < 5) return form2;
  if (n1 == 1) return form1;
  return form5;
}

currentSegmentText = String(days) + " " + plural_ru(days, "день", "дня", "дней");

sprintf(buf, "%ld %s", hours, plural_ru(hours, "час", "часа", "часов").c_str());

sprintf(buf, "%ld %s", minutes, plural_ru(minutes, "минута", "минуты", "минут").c_str());

sprintf(secondsBuf, "%ld %s", currentSecond, plural_ru(currentSecond, "секунда", "секунды", "секунд").c_str());

single-line countdown

sprintf(buf, "%s: %ld %s %ld %s %ld %s %ld %s", 
        label.c_str(),
        days, plural_ru(days, "день", "дня", "дней").c_str(),
        hours, plural_ru(hours, "час", "часа", "часов").c_str(),
        minutes, plural_ru(minutes, "минута", "минуты", "минут").c_str(),
        seconds, plural_ru(seconds, "секунда", "секунды", "секунд").c_str());

I don't know how useful this information is, I only managed to display the days of the week and month. The countdown couldn't be done in Russian. Cyrillic glyphs were displayed incorrectly.

mfactory-osaka, 2025-09-26:
Added Russain today.

days of the week, month and weather description are translated.

переменная облачность will display as PEREMENNAYA OBLACNOST

countdown is not translated in any laguage

days will be: "pn", "vt", "sr", "ch", "pt", "sb", "vs"
months will be: ian", "feb", "mar", "apr", "mai", "iun", "iul", "aug", "sep", "oct", "noi", "dec"

let me know if it works

closed 2025-09-26 by mfactory-osaka

Mart-Black, 2025-09-26:
It's working! Thanks!
But in the code in all languages, Sunday comes first. The update in Russian is currently set to Monday, so the days of the week are displayed with an offset.
I also played with the letters a little, in this version it looks more like the Russian abbreviations of the days of the week to me. I've been playing with it for months, and I'll add more later, as it became clearer to a native speaker.
I also didn't find anything adequate for Thursday. In Russian, this is "чт" in the font, the number and the letter "4t" are ideally suited. But the upper strip is erased at the same time.
I still couldn't find the right letter for subota. The "G" is similar to the Russian "Б", but because of the rounded corners, it feels a bit wrong.
Thanks again!
{ "ru", { "b&c", "n&h", "b&t", "c&p", "4&t", "n&t", "c&b" } },
So Russian abbreviations and what did I get
Sunday-ВС
Monday-ПН
Tuesday-ВТ
Wednesday-СР
Thursday-ЧТ-4Т
Friday-ПТ
Saturday-СБ-СВ

Mart-Black, 2025-10-08:
Good afternoon. the days of the week are displayed perfectly. Is it possible to add some letters for the months? Should I try to find English letters for months, as if they were in Russian?